korginc/logue-sdk #4: ./logue-cli probe not working

sqpierce, 2018-06-28:
**Describe the bug**
When I run ```./logue-cli probe``` from command line (Mac OS X) with the Prologue connected, I get an error.

```
Error: API version request timed out.
Error: Prologue handshake failed.
```

**To Reproduce**
Steps to reproduce the behavior:
```
cd logue-sdk/tools/logue-cli
./get_logue_cli_osx.sh
./logue-cli probe
```

**Expected behavior**
I'm not sure exactly what's supposed to happen, but I would presume it would not be 'Error'.

**Screenshots**
https://github.com/sqpierce/logue-sdk/blob/master/screenshot1.png

[image]

**Desktop (please complete the following information):**
 - OS: Mac OS X
 - Version 10.13.4 (High Sierra)

**Additional context**
I believe I have all the dependencies set up. I was able to compile my program, and when doing ```./logue-cli check -u /path/to/my/unit``` I get a good result ```> Parsing unit archive
> Parsing manifest
> Parsing unit binary payload
> Looks okay.```

Is the prologue meant to be started in a special mode for this to work? (i.e. hold down 6 and or 8 as when doing software updates?)

etienne-korg, 2019-06-20:
If the ports are not automatically detected the following should be used:
Input: prologue KBD/KNOB
Output: prologue SOUND

The other two refer to the legacy MIDI ports on the back of the prologue.

gekart, 2019-07-07:
Tried that with the current version  0.06-0b, doesn't work. Same problem as above.

etienne-korg, 2019-07-09:
There's a pending update that should fix a dynamic linking issue on OSX. 
Until it gets officially uploaded here's a direct download link:
https://www.dropbox.com/s/kyjexw6l6ydxzi7/logue-cli-osx-0.07-0b.zip?dl=0

gekart, 2019-07-09:
Yay! Version 0.07 works flawlessly on my XD.

gekart, 2019-07-16:
@etienne-korg I included it into VSCode https://www.youtube.com/watch?v=JqyhttRtbnI

etienne-korg, 2019-09-18:
version v0.07-2b is now available via the download scripts so closing this issue.

closed 2019-09-18 by etienne-korg