korginc/logue-sdk #6: missing dependencies for compiling Waves on OSX El Capitan v10.11.6

phillipsjus, 2019-02-02:
**Describe the bug**
I havean iMac running OSX El Capitan v10.11.6.  I followed instructions in https://github.com/korginc/logue-sdk/blob/master/platform/prologue/README.md under Building a Project section including all the instructions for setting up the toolchain.  I went into the logue-sdk/platform/prologue/demos/waves directory and ran make.  The following error

Justins-iMac:waves justin$ make
Compiling _unit.c
In file included from ../../inc/utils/fixed_math.h:47:0,
                 from ../../inc/osc_api.h:46,
                 from ./inc/userosc.h:47,
                 from ./tpl/_unit.c:42:
../../inc/utils/cortexm4.h:45:31: fatal error: arm_math.h: No such file or directory
compilation terminated.
make: *** [build/obj/_unit.o] Error 1

**To Reproduce**
Steps to reproduce the behavior:
1. clone logue-sdk
2. follow instructions for setting up toolchain for OSX
3. from logue-sdk/platform/prologue/demos/waves directory run 'make'

**Expected behavior**
results in compilation error 
In file included from ../../inc/utils/fixed_math.h:47:0,
                 from ../../inc/osc_api.h:46,
                 from ./inc/userosc.h:47,
                 from ./tpl/_unit.c:42:
../../inc/utils/cortexm4.h:45:31: fatal error: arm_math.h: No such file or directory
compilation terminated.
make: *** [build/obj/_unit.o] Error 1

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: OSX El Capitan v10.11.6.
 - Version: El Capitan v10.11.6.

distropolis, 2019-02-12:
I can confirm this in Mac OS 10.13.6 as well. Same issue.

distropolis, 2019-02-12:
I was able to compile the test project after pulling the following files from the Windows ARM archive listed in the /tools section:

arm_math.h
core_cmInstr.h
cmsis_gcc.h
core_cmFunc.h
core_cmSimd.h

I placed them in the ./tools/gcc/gcc-arm-none-eabi-5_4-2016q3/arm-none-eabi/include/ directory of my project.

This compiles with the following message:

Arman-Desk-iMac:square arman-desk-iMac$ make
Compiling _unit.c
Compiling square.cpp
Linking ../../build/square_test.elf
Creating ../../build/square_test.hex
Creating ../../build/square_test.bin
Creating ../../build/square_test.dmp

   text	   data	    bss	    dec	    hex	filename
    544	      0	     28	    572	    23c	../../build/square_test.elf

Creating ../../build/square_test.list
Packaging to ./square_test.mnlgxdunit

Done

etienne-korg, 2019-02-25:
The _arm_math.h_ header is provided by CMSIS which is a submodule of the logue-sdk repository.
It should be located here: platform/ext/CMSIS/CMSIS/Include/arm_math.h

Make sure you run _git submodule update --init_ before attempting to build projects.

closed 2019-03-01 by etienne-korg