korginc/logue-sdk #7: Cannot compile (missing CMSIS arm_math.h)

corinrose, 2019-03-13:
**Describe the bug**
Trying to compile the demo code in demos/waves gives the following output:

~~~
:waves > make
Compiler Options
../../../../tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mno-thumb-interwork -DTHUMB_NO_INTERWORKING -DTHUMB_PRESENT -g -Os -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -fcheck-new -std=c11 -mstructure-size-boundary=8 -W -Wall -Wextra -Wa,-alms=./build/lst/ -DSTM32F401xC -DCORTEX_USE_FPU=TRUE -DARM_MATH_CM4 -D__FPU_PRESENT -I. -I./inc -I./inc/api -I../../inc -I../../inc/dsp -I../../inc/utils -I../../../ext/CMSIS/CMSIS/Include

Compiling _unit.c
In file included from ../../inc/utils/fixed_math.h:47:0,
                 from ../../inc/osc_api.h:46,
                 from ./inc/userosc.h:47,
                 from ./tpl/_unit.c:42:
../../inc/utils/cortexm4.h:45:31: fatal error: arm_math.h: No such file or directory
compilation terminated.
make: *** [Makefile:189: build/obj/_unit.o] Error 1
~~~

This is on Linux, but I have the exact same problem on a friends Mac computer.

**To Reproduce**
Steps to reproduce the behavior:
1. Install tools
2. Go to demo code, run make

**Expected behavior**

Compilation

**Desktop (please complete the following information):**
 - OS: Linux 5 and  Mac os Mojave
 - Version Downloaded on March 13th

damnatron, 2019-03-14:
I had the same issue on Windows. See closed issue https://github.com/korginc/logue-sdk/issues/6 

"The arm_math.h header is provided by CMSIS which is a submodule of the logue-sdk repository.
It should be located here: platform/ext/CMSIS/CMSIS/Include/arm_math.h

Make sure you run git submodule update --init before attempting to build projects."

That fixed it for me.

RikMaxSpeed, 2019-05-25:
"git submodule update --init" fixed it for me too - thanks!
Would be good if this were mentioned in the ReadMe.

etienne-korg, 2019-06-20:
Making sure submodules are initialized and updated is mentioned in the platform's README but could indeed be made clearer.

etienne-korg, 2019-06-20:
I added a troubleshooting item regarding this on the platform README.

closed 2019-06-20 by etienne-korg