korginc/logue-sdk #12: `user_osc_param` does not expose velocity data

jakebromberg, 2019-08-21:
**Is your feature request related to a problem? Please describe.**
I would like to modulate my oscillators based on velocity input. Currently this is unavailable via the `user_osc_param`, which would seem like a likely place to pull this from. 

**Describe the solution you'd like**
I do see that the `user_osc_param` has an [extra three bytes](https://github.com/korginc/logue-sdk/blob/master/platform/minilogue-xd/osc/inc/userosc.h#L59) of padding. Allocating one of those bytes for velocity measurements seems like a reasonable solution.

etienne-korg, 2019-09-03:
Velocity is currently not available in the oscillator runtime, but request noted.

ptarabbia, 2019-12-28:
Yes please ! That would be extremely useful

seanwayland, 2022-10-16:
This would be so useful. All my years of piano practice and keyboard practice could be put to good use !