korginc/logue-sdk #21: Extra parameters for effects

ptarabbia, 2019-12-28:
**Is your feature request related to a problem? Please describe.**
The API and json files currently support extra parameters for effects, just like custom oscillators, but these can't be accessed through the front panel (that I know of).

**Describe the solution you'd like**
(this is for the minilogue XD - hopefully it can translate easily to the other hardware units).
In "Program Edit" mode, have button 11 (currently inactive) access parameters for whatever custom delay is selected by the "Mod/Del/Rev" switch. Navigation would work exactly like button 10 (for custom oscillator parameters).

dukesrg, 2019-12-29:
That could be even more  useful for NTS-1 as it can be used as a standalone universal FX box for external audio input

misterinterrupt, 2020-02-24:
I think this is essential! So much potential to create and develop unique and interesting effects on the NTS-1. Cheers!