korginc/logue-sdk #22: Debugging features

ptarabbia, 2019-12-29:
**Is your feature request related to a problem? Please describe.**
It's really hard to investigate what's going on when things don't play the way I'd expect

**Describe the solution you'd like**
A method allowing to print a string to the OLED display would be ideal for debugging purposes.

gazzar, 2019-12-29:
> 
> 
> **Is your feature request related to a problem? Please describe.**
> It's really hard to investigate what's going on when things don't play the way I'd expect
> 
> **Describe the solution you'd like**
> A method allowing to print a string to the OLED display would be ideal for debugging purposes.

or even printing to the 7-segment display might often be better since you will be using the OLED display for setting parameters.

ptarabbia, 2019-12-29:
> or even printing to the 7-segment display might often be better since you will be using the OLED display for setting parameters.

Yes. Anything that can be used to display something :)

gazzar, 2020-01-01:
Hey @ptarabbia I managed to get the Korg waves demo oscillator compiled with web assembly using the alpha/wasm-builds branch code and following the documentation there. I can now add standard string output to the oscillator code with printf or std::cout, load it in the browser simulator, and it outputs strings in the Chrome debug console. It's probably not relevant here, but the documentation says it is untested in Windows Subsystem for Linux but I had no problem with that. I'm also successfully building oscillators with gcc under WSL.

smith-xyz, 2020-03-07:
Is there any success in configuring a way to build and debug in vs code?  I'll see about looking into alpha/wasm-builds branch to load it in the browser simulator but was just curious on any attempts to set this up in vs code.