korginc/logue-sdk #23: Customize display of parameter values

ptarabbia, 2019-12-29:
**Is your feature request related to a problem? Please describe.**
The manifest.json file is used to specify ranges for the parameters, but there is no way to customize how these display on the OLED. For example a parameter with 3 values for setting a LFO shape to Sine, Square and Triangle, will display as "1", "2", "3".

**Describe the solution you'd like**
Add a new hook method:
void _hook_display(uint16_t index, uint16_t value, char* display);
The custom osc/fx would write in the provided "display" buffer string, which text to show when parameter 'index' has value 'value'.

ptarabbia, 2019-12-29:
by the way, the current display of parameters is off by one. If you have a parameter entry such as:
["PlayMode", 1, 4,  ""]
it will display as 2..5 in the OLED.

dukesrg, 2019-12-29:
> by the way, the current display of parameters is off by one.
This is an expected behaviour:
https://github.com/korginc/logue-sdk/blame/master/platform/prologue/README.md#L129