korginc/logue-sdk #25: Negative parameter values displayed as "999"

dukesrg, 2019-12-31:
**Describe the bug**
Negative parameter values for both percent and typeless values displayed as "999" at leas at NTS-1

**To Reproduce**
Steps to reproduce the behavior:
1. Make custom oscillators with parameter of percent type from -100 to +100
2. Set parameter knob to anything less than 12 o'clock
3. observe '999' displayed while listen that sound actually reflects parameter changes.

**Expected behavior**
Should correctly display -100...100 at least for percent type values according to SDK manual.

**Screenshots**
If applicable, add screenshots to help explain your problem.

dj-turkish-ice-cream, 2020-01-14:
I also repro'd this with both a typeless and a "%" type parameter with range -12, 12.

In the case of a typeless parameter, value "-999" is shown from minimum knob position to 12 o'clock. After that, values from 0 to 12 are shown. The -999 range seems to behave the same as value 0.

If the type is "%", the behavior is similar except the value shown in the negative range is "999" (no minus sign).

etienne-korg, 2020-01-27:
Will be fixed in a future update.