korginc/logue-sdk #30: OSC_NOTEOFF not invoked until all keys are released

dj-turkish-ice-cream, 2020-01-14:
OSC_NOTEOFF documentation says it is "Called whenever a note off event occurs." This is a bit vague, but arguably the behavior should mirror midi note off events.

When multiple keys are held down and released one-by-one, the release of each key produces a midi note off event. However, when these midi messages are passed to the NTK1, OSC_NOTEOFF is only invoked when the last held note is released.

**To Reproduce**
I used [sendmidi](https://github.com/gbevin/SendMIDI) to send the midi commands below.

1. Replace the OSC_NOTEOFF code of the Waves oscillator to turn on bit crush:
```
  s_waves.params.bitcrush = clip01f(1.5f);
  s_waves.state.flags |= Waves::k_flag_bitcrush;
```
2. Compile, load and select oscillator.
3. Send midi note on 50. Sound starts as expected.
4. Send midi note on 51. Pitch goes up one semitone as expected.
5. Send midi note off 51.
Expected: Pitch goes down one semitone, bitcrush is enabled.
Actual: Pitch goes down one semitone, bitcrush is not enabled - i.e. OSC_NOTEOFF is not invoked.

6. Send midi note off 50. Sound stops as expected. Bitcrush is enabled (can be verified by setting EG to Open).

etienne-korg, 2020-01-27:
The NTS-1 handles MIDI notes in legato mode.
Will interpret this as a request for non-legato mode.

dukesrg, 2020-01-27:
@etienne-korg Passing user_osc_param_t in NOTE_ON/NOTE_OFF and in OSC_CYCLE is really confusing, you need to clarify the behaviour in current documentation. 

I suspect #26 is of the similar nature.