korginc/logue-sdk #31: Cant Compile

simi0321, 2020-01-24:
I ve just set up the project accordingly but receive an error when I try to compile the demo:
"user@pc:~/prologue_sdk/logue-sdk/platform/prologue/demos/waves$ make
Compiler Options
../../../../tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc -c -mcpu=cortex-m4 -mthumb -mno-thumb-interwork -DTHUMB_NO_INTERWORKING -DTHUMB_PRESENT -g -Os -mlittle-endian -mfloat-abi=hard -mfpu=fpv4-sp-d16 -fsingle-precision-constant -fcheck-new -std=c11 -mstructure-size-boundary=8 -W -Wall -Wextra -Wa,-alms=./build/lst/ -DSTM32F401xC -DCORTEX_USE_FPU=TRUE -DARM_MATH_CM4 -D__FPU_PRESENT -I. -I./inc -I./inc/api -I../../inc -I../../inc/dsp -I../../inc/utils -I../../../ext/CMSIS/CMSIS/Include

Compiling _unit.c
make: ../../../../tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: Command not found
Makefile:190: recipe for target 'build/obj/_unit.o' failed
make: *** [build/obj/_unit.o] Error 127
"
I checked if its a simple path-missmatch but the file is there. I use an up to date Ubuntu 16.04 sytem.

tokuhira, 2020-01-25:
Is "GNU Arm Embedded Toolchain" already installed?
https://github.com/korginc/logue-sdk/tree/master/tools/gcc

simi0321, 2020-01-27:
> Is "GNU Arm Embedded Toolchain" already installed?
> https://github.com/korginc/logue-sdk/tree/master/tools/gcc

Yes, it is. I used the provided script to do so.

tokuhira, 2020-01-27:
Which version of make do you use? My version is shown below. macOS but ... 

$ make -v
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0

etienne-korg, 2020-01-27:
Tested a fresh clone > submodule init/update > get gcc > make waves demo on  a clean Ubuntu VM (19.10) and it built fine. 
Perhaps have a look at `make --trace` or `make -d` for more details as to what is happening. Also, perhaps reset/stash any local changes you have (if any).

simi0321, 2020-01-28:
Hi Guys, thx for the quick replies and the help. I could solve it! The issue is that one should probably **not** use the provided script to install gcc but simply download and unpack it on your own. The script installs a 32bit version despite 64bit architecture and just notifies you shortly in the script that they do something for 32bit. I replaced the gcc with the 64 bit binarys (from https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads) and could compile without errors. However, I have not (yet) tested the 64 toolchain properly. If the 64 bit works well for other people maybe it can be added to the docu for the gcc part? In any case I would suggest to emphasis it that you should not use the install script with 64 bit. Maybe renaming it to "get_gcc_linux_32bit_only.sh" helps already.
Cheers

closed 2020-01-28 by simi0321

etienne-korg, 2020-01-28:
The 32bit toolchain wors fine on the x86_64 ubuntu installs I have at hand, but it is indeed possible that it won't on some specific installations.
Note that the script already outputs a warning that says it assumes a 32bit install.

tokuhira, 2020-02-06:
The conclusion is already made, just for your information. This issue was reproduced on a clean install of my Ubuntu 18.04 LTS (x86_64). However, by adding the "libc6-i386" package, the required version of the 32-bit prebuild compiler worked.

apiel, 2021-03-04:
Thx for the hind, actually seem that on ubuntu you can install it directly from the package manager with `apt-get install gcc-arm-none-eabi`