korginc/logue-sdk #32: User facing osc-specific parameters are not exposed to MIDI

ggsubs, 2020-01-26:
User facing osc-specific parameters are not transmitted to MIDI and either changeable with control change messages. Given Shape and ShiftShape is sent/received from MIDI, developer expectations are to include the user messages too.

john-k-walton, 2020-01-29:
It could be useful to have six/eight midi CC's to control osc params outboard instead of diving menus. Exposing EG and LFO controls directly during performances is a noble feature,

dukesrg, 2020-01-29:
Minilogue XD have NRPMs for all six osc parameters. I wonder why it is not implemented in NTS-1.

leandrob13, 2023-12-26:
Crossing our fingers so that user osc params get MIDI control in 2024.

john-k-walton, 2023-12-26:
yeah, I've given up on getting any Prologue functional updates. I use a Yorick LFE; a low frequency external modulator with LFO's, EG's, sending note velocity back to Shape, or Shift-Shape, and rewrite my code to put the best stuff on those controls. much of the stuff I want is actually in there already, I just have to use the (freely available) memory map for the all the parameters to access them for this firmware.