korginc/logue-sdk #33: Knob controls being sent to incorrect FX module (delfx and revfx)

miz884, 2020-01-31:
**Describe the bug**
Even if the FX mode is switched from delfx to revfx, the knob values is being sent to delfx, and vice-versa.

**To Reproduce**
Steps to reproduce the behavior:
1. Push DELAY button to set the FX mode to delfx.
2. Select any delay module by TYPE knob.
3. Use A or B knob to control the module.
4. Push REVERB button to switch the FX mode to revfx.
5. Without touching TYPE knob, twist the A or B knob.

(Issue) Then, the knob controls will be sent to the delfx module, although the FX mode is in revfx currently (due to the step 4 above).

6. Select any reverb module by TYPE knob.

Then, the knob controls will start to be sent to the correct revfx module.

**Expected behavior**
The knob controls should be sent  to the selected modules.

**Screenshots**
N/A

**Desktop (please complete the following information):**
 - OS:
$ sw_vers
ProductName:    Mac OS X
ProductVersion: 10.15.2
BuildVersion:   19C57

 - Version
$ ./logue-cli probe -i 0 -o 0
> Device: nutekt digital
> System version: 1.03
> Logue API version: 1.01-0
> Available modules:

**Additional context**
N/A

miz884, 2020-01-31:
This issue is not reproducible on MODFX.

etienne-korg, 2020-02-04:
Hi, thanks for reporting this issue, however it is not related to the logue-sdk but rather to the NTS-1 itself. This has already been reported elsewhere and will be addressed in a future update. However product related issues will not be tracked on this repository.

Product-related issues should be reported through the appropriate customer support channels.

closed 2020-02-04 by etienne-korg