korginc/logue-sdk #38: Demos for the effects?

schollz, 2020-05-04:
This is a really amazing resource. The NTS-1 is the kind of synthesizer I've always dreamed of, being able to program with a really well developed framework is fantastic.

I found all the instructions useful for cloning the repo, installing the toolchain, etc. I found that the demo for the waves is absolutely crucial for me to get going on programming difference oscillators.

Is there any demo for the effects? I'm interested in making a delay. The [template folder](https://github.com/korginc/logue-sdk/tree/master/platform/nutekt-digital/delfx) for the effect is a crucial starting point, but [the included code](https://github.com/korginc/logue-sdk/blob/master/platform/nutekt-digital/delfx/tpl/_unit.c) is still a little too bare bones for me to tinker with.

Please let me know if this repo has any effects demos I missed, or if anyone else out there has effects demos that they could share! I did find [@dukesrg's fork with cool effects](https://github.com/dukesrg/logue-fx) but it looks as if its deviated quite a bit from the original template in this repo.

dukesrg, 2020-05-04:
@schollz 
1. Read throughfully https://github.com/korginc/logue-sdk/tree/master/platform/nutekt-digital
2. Have a look at https://github.com/korginc/logue-sdk/blob/master/platform/nutekt-digital/delfx/tests/src

All you have to do is to implement several API fuctions, just don't bщther with the rest of the project template!
One of the reason of my logue-sdk fork to reduce the project footprint and move away the stock code and hardware abstraction  which noone ever shoud touch.

schollz, 2020-05-05:
@dukesrg Thanks so much for the quick response. I missed the code in those tests somehow, I think I'll be able to manage with that!

closed 2020-05-05 by schollz