korginc/logue-sdk #40: Shape LFO parameter: NTS1 and minilogue xd

boochow, 2020-05-05:
**Describe the bug**
NTS-1 and minilogue xd give different values for shape_lfo parameter in OSC_CYCLE.

**To Reproduce**
Build square osc in the osc/tests/square and install it to NTS-1 and minilogue xd.
Set LFO rate to slow ( < 1Hz), LFO target to pitch, and maximize intensity.
Hit any key and hear how the pitch changes. NTS-1's pitch never goes down under the key while minilogue xd goes.

**Expected behavior**
I think minilogue xd's behavior is more preferred.

**Screenshots**
The attached image shows how PWM changes with LFO. In both cases, the intensity is the max value.
[image]

dukesrg, 2020-05-05:
@boochow At least on NTS-1, shape_lfo parameter of OSC_CYCLE only has a non-zeo value when LFO target is SHAPE. When LFO target is PITCH, the params->pitch itself is passed modulated.
You can clearly see both osc/tests/square.c have exactly the same source code for moth NTS-1 and Minilogue XD and shape_lfo does not affect pitch here.

I did not find a word about neither pitch LFO sensitivity nor pitch bend sensitivity in both NTS-1 and Minilogue XD manuals. MIDI implementation in those Korg products is... ugh... very ascetic :)

boochow, 2020-05-05:
@dukesrg Yes, the same source, different results. That is the problem. I have been using NTS-1 and got a minilogue xd today and found this issue. I suppose it may not be an issue of the SDK but the wrong behavior of NTS-1.

dukesrg, 2020-05-05:
@boochow to clearify: you're experiencing a different pitch LFO depth applied with the same pitch LFO depth value on Minilogue XD and NTS-1.
This is not about params->shape_lfo so but about different internal implementation in hardware.

boochow, 2020-05-05:
The same settings for different hardware cause different results. I couldn't clarify where the problem actually is, but anyway, I had to write different codes for different hardware to get the same results.

dukesrg, 2020-05-05:
@boochow Did the test, and
Did a test: NTS-1 have triangle wave in 0.0...1.0 q31 value range (there is an offset in the oscilloscope app, that is a positive-only triangle!)
[image]
This is for SHAPE LFO target, looks like there is a typo in the first post stating pith as a target.

boochow, 2020-05-05:
Thank you for testing. However, it is not a typo. I saw both targeting pitch and targeting shape seemed to have the same problem: NTS-1 sends an OSC module LFO values from 0.0 to 1.0 while minilogue xd sends values from -1.0 to 1.0.
In case of shape_lfo parameter, it is easy to get the same results from NTS1 and minilogue xd by replacing
NTS-1: ` q31_to_f32(p->shape_lfo) `
to
minilogue xd: ` (q31_to_f32(p->shape_lfo) / 2 + 0.5)`.

The pitch problem is more serious because `OSC_CYCLE()` receives the LFO-modulated pitch, and it is difficult to know what the actual LFO value is.

dukesrg, 2020-05-05:
@boochow Thanks, now it makes sense. Really looks like a bug in NTS-1 runtime.
Since there is no way to get the firmware version, I'd stick to correct Minilogue XD behaviour. Unless maintainers confirm this is an expacted behaviour for NTS-1. Let's hope there will be a bugfix.

dukesrg, 2020-05-07:
@etienne-korg please comment on this https://github.com/korginc/logue-sdk/blob/alpha/wasm-builds/platform/ext/WAB/cpp/logue_wrapper.cpp#L96 - shape LFO supposed to be in [0...1] range  in OSC_CYCLE

boochow, 2020-06-06:
> NTS-1: `q31_to_f32(p->shape_lfo)`
> to
> minilogue xd: ` (q31_to_f32(p->shape_lfo) / 2 + 0.5)`.

I found that the code above still has an issue. When the LFO intensity is zero, the shape_lfo value becomes zero for both NTS-1 and minilogue xd. However, the above code for minilogue xd returns 0.5 when the shape_lfo value is zero. To get the same result with NTS-1 on minilogue xd, we need the LFO intensity value.