korginc/logue-sdk #42: Issues with float_math.h

musicalentropy, 2020-06-15:
In float_math.h, I have seen a few issues: 

- some typos with "ampltitude" which should be replaced with "amplitude"
- in the function fastlogf and fasterlogf, the constant "0.69314718f" should be replaced with the macro MN_LN2
- the function "fasterampdbf" might be even faster if the factor is precalculated instead of being 20.f/fasterlog2f(10)
- the function "fasterdbampf" is wrong, it should be "fasterpowf(10.f, 0.05f*db)" instead of "20.f*fasterpowf(10.f, 0.05f*db)"

etienne-korg, 2020-06-17:
Thanks for pointing this out.
Fixed in ab654af0e29e5eb6eab7289d05699b45417d1f68

closed 2020-06-17 by etienne-korg

musicalentropy, 2020-06-17:
Thanks! And don't forget the "/** dB to ampltitude" typos, like for example in line 796 ;)