korginc/logue-sdk #43: Attempting to probe minilogue XD through logue-sdk fails [Linux]

serratedserenade, 2020-09-06:
**Describe the bug**
When I try to probe my minilogue through the cli it produces either "Search device request timed out" or "API version request timed out"

**To Reproduce**
Here is my terminal output:
```
✘ adamdc@AdamHome  ~/Documents/loguesdk modules/KorgCorrosion102/minilogue xd  logue-cli probe
Error: Could not find matching logue MIDI ports.

Logue interface connection failed.
  Available MIDI inputs:
    in  0: Midi Through:Midi Through Port-0 14:0
    in  1: minilogue xd:minilogue xd MIDI 1 32:0
    in  2: minilogue xd:minilogue xd MIDI 2 32:1

  Available MIDI ouputs:
    out 0: Midi Through:Midi Through Port-0 14:0
    out 1: minilogue xd:minilogue xd MIDI 1 32:0
    out 2: minilogue xd:minilogue xd MIDI 2 32:1

 ✘ adamdc@AdamHome  ~/Documents/loguesdk modules/KorgCorrosion102/minilogue xd  logue-cli probe -i 1 -o 1
Error: Search device request timed out.
Logue handshake failed.
 ✘ adamdc@AdamHome  ~/Documents/loguesdk modules/KorgCorrosion102/minilogue xd  logue-cli probe -i 2 -o 2
Error: API version request timed out.
Logue handshake failed.
```

**Expected behavior**
Able to probe my minilogue xd

**Desktop (please complete the following information):**
Arch Linux 5.8.5-arch1-1

msil, 2020-10-22:
For what it's worth I am seeing exactly the same issue with my minilogue xd and Ubuntu 20.04 (kernel 5.4)

Running latest firmware (v2.10)

volca02, 2021-02-13:
I am encountering the exact same behavior on arch linux with kernel 5.10. It would seem minilogue does not get the second sysex message or does not respond to it.

I tried using a usb to midi cable, which yielded the same result.

```
./logue-cli probe -v -d -i 2 -o 2

>>> { f0, 42, 50, 0, 55, f7 }
<<< { f0, 42, 50, 1, 8, 55, 51, 1, 0, 0, 2, 0, 1, 0, f7 }
> Device: minilogue xd
> System version: 2.01
>>> { f0, 42, 30, 0, 1, 51, 17, f7 }
Error: API version request timed out.
Logue handshake failed.
```

 I'd wager this is a problem of the message not being composed well in the first place (can't be sure here, but the third byte differs in my case compared to the report above) or alsa changing the midi message content or outright dropping it.

Either way, after running the logue-cli exactly once on new connection, this is the result:

```shell
cat /proc/asound/xd/midi0 
minilogue xd

Output 0
  Tx bytes     : 0
Output 1
  Tx bytes     : 14
Input 0
  Rx bytes     : 0
Input 1
  Rx bytes     : 64
```

The number of bytes received does not match, which is probably okay due to midi clock being sent on the midi out of the minilogue all the time.

volca02, 2021-02-14:
Okay after digging through this, the issue is simple - `logue-cli` does not handle MIDI Channel well.

The API version request does not use the correct midi channel - the first response from minilogue reports port no 9 (hex 8) clearly, yet the second request uses midi channel 1 (hex 0) instead. (As per Korg's own MIDI implementation document, the third byte should be `3g`, where g is the midi channel)

Changing the midi channel on minilogue to 1 fixes the issue.

```
>>> { f0, 42, 50, 0, 55, f7 }
<<< { f0, 42, 50, 1, 0, 55, 51, 1, 0, 0, 2, 0, 1, 0, f7 }
> Device: minilogue xd
> System version: 2.01
>>> { f0, 42, 30, 0, 1, 51, 17, f7 }
<<< { f0, 42, 30, 0, 1, 51, 47, 2, 1, 1, 0, f7 }
> Logue API version: 1.01-0
> Available modules:

>>> { f0, 42, 30, 0, 1, 51, 18, 0, f7 }
<<< { f0, 42, 30, 0, 1, 51, 48, 0, f7 }
Global: no info.
>>> { f0, 42, 30, 0, 1, 51, 18, 1, f7 }
<<< { f0, 42, 30, 0, 1, 51, 48, 1, 0, 1, 74, 1f, 0, 0, 0, 18, 0, 0, 0, 10, f7 }
Modulation FX: [ slot_count: 16, max_payload_size: 8180, max_load_size: 6144 ]
>>> { f0, 42, 30, 0, 1, 51, 18, 2, f7 }
<<< { f0, 42, 30, 0, 1, 51, 48, 2, 0, 1, 70, 3f, 0, 0, 0, 30, 0, 0, 0, 8, f7 }
Delay FX: [ slot_count: 8, max_payload_size: 16368, max_load_size: 12288 ]
>>> { f0, 42, 30, 0, 1, 51, 18, 3, f7 }
<<< { f0, 42, 30, 0, 1, 51, 48, 3, 0, 1, 70, 3f, 0, 0, 0, 30, 0, 0, 0, 8, f7 }
Reverb FX: [ slot_count: 8, max_payload_size: 16368, max_load_size: 12288 ]
>>> { f0, 42, 30, 0, 1, 51, 18, 4, f7 }
<<< { f0, 42, 30, 0, 1, 51, 48, 4, 0, 23, 70, f, 0, 0, 0, 0, 0, 0, 0, 10, f7 }
Oscillator: [ slot_count: 16, max_payload_size: 36848, max_load_size: 32768 ]

... and so on...
```

The same problem happens with logue-cli.exe (tried that by using it via wine).

Please, Korg, fix the logue-cli utility so it is usable on a different channel than no. 1.

rafaelpernil2, 2021-04-21:
I can confirm. I was having issues with logue-cli using MIDI channel 2. Thanks for the input, @volca02  and I hope KORG fixes this issue.

closed 2023-02-11 by serratedserenade

znmeb, 2023-02-11:
This might explain some of the issues I've been having with a completely different use case: sending user scales to the Minilogue XD via SysEx from the ODDSound MTS-ESP suite. Maybe I need to specify channel 1 and make sure the Minilogue XD is set to listen on channel 1.