korginc/logue-sdk #45: unable to compile waves demo on WLS

markoparkermarsenic, 2021-01-16:
**Describe the bug**
When attempting to `make` waves for the XD the following error occurs:

`Compiling _unit.c
../../../../tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: 1: ../../../../tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: Syntax error: word unexpected (expecting ")")
Makefile:190: recipe for target 'build/obj/_unit.o' failed
make: *** [build/obj/_unit.o] Error 2
`

I am running WSL with my loge-SDK repo inside `D:\` the the SKD is a submodule of another project

*To Reproduce**
Steps to reproduce the behavior:
1. `make` wthin the waves dir

**Expected behavior**
successful compilation 

**Desktop:**
 - Windows 10: 
 - WSL v.1

**Additional context**
gcc works

tokuhira, 2021-01-16:
FYI, hope helps your situation:

https://github.com/korginc/logue-sdk/issues/31#issuecomment-582836208

gazzar, 2021-02-01:
This is what worked for me last time I tried it for getting the logue-sdk working in wsl. I haven't tried since I switched over to wsl2. Hope this helps.

```
	1. Install wsl and the Ubuntu image from Windows Store
	2. Run wsl
	3. I want to use vscode. I have it installed in Windows already. Run vscode from Linux commandline
	$ code
	## This runs VSCode and prompts to install the "remote wsl" extension in VSCode which I did
	4. Install qemu
	## This is necessary to be able to run ELF 32-bit LSB executables, of which the arm gcc compiler tools are an example.
	
	## https://stackoverflow.com/a/49405605
	$ sudo apt install qemu-user-static
	$ sudo update-binfmts --install i386 /usr/bin/qemu-i386-static --magic '\x7fELF\x01\x01\x01\x03\x00\x00\x00\x00\x00\x00\x00\x00\x03\x00\x03\x00\x01\x00\x00\x00' --mask '\xff\xff\xff\xff\xff\xff\xff\xfc\xff\xff\xff\xff\xff\xff\xff\xff\xf8\xff\xff\xff\xff\xff\xff\xff'
	
	# Add the following line to .bashrc:
	sudo service binfmt-support start
	
	$ source .bashrc
	
	$ sudo dpkg --add-architecture i386
$ sudo apt update
	
	5. Follow the instructions in tools/gcc/README.md
	$ cd /mnt/d/
	$ git clone https://github.com/korginc/logue-sdk.git
	$ cd logue-sdk
	$ git submodule update --init
	$ ./get_gcc_linux.sh
	
	# Check if it will work
	$ file tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc
	
	## The following output indicates that the gcc compiler is an ELF 32-bit LSB executable as it should.
	tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/arm-none-eabi-gcc: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-, for GNU/Linux 2.6.8, stripped
	
	$ cd platform/minilogue-xd/demos/waves/
	$ make

```

augusto, 2022-09-14:
For ubuntu 20 on WSL I also had to install the package libc6-i386. 
```
sudo apt install libc6-i386
```