korginc/logue-sdk #55: Open the sources of logue-cli

Noir-, 2021-07-12:
**Is your feature request related to a problem? Please describe.**
It looks like you could need some help with polishing things, especially on Linux where #37 is a show stopper at the moment.

**Describe the solution you'd like**
If you open the sources, things like this will be quickly sorted out by the community. It looks like you don't have much to lose because the MIDI communication is easily sniffable anyway.

jpnielsen, 2023-09-04:
Yes, Please open logue-cli

Is anyone from KORG listening ?

MKoussa, 2025-01-01:
Please, Korg!

I just spent 2 hours trying to figure out why I couldn't move an fx over and it was because the midi buffer was set too low.

I had to manually transfer the sysex data...