korginc/logue-sdk #58: MIDI messages callbacks value resolution increase for dtumlogue SDK

dukesrg, 2022-11-10:
To avoid API changes in future it make sense to increase the resolution for the values passed to the callbacks in this section:
https://github.com/korginc/logue-sdk/blame/master/platform/drumlogue/README.md#L319 in compliance with MIDI 2.0 specs:
Velocity should be 16-bit
Pitch bend, Channel pressure and Aftertouch - all 32-bit

dukesrg, 2022-11-15:
-

etienne-korg, 2022-11-15:
drumlogue is not MIDI 2.0 compliant.

If support for MIDI 2.0 is added in the future, or a compliant product uses an API based on the drumlogue's, then optional callbacks aligned with MIDI 2.0 requirements will be added.
The current API will remain aligned with MIDI 1.0 requirements.

closed 2022-11-15 by etienne-korg