korginc/logue-sdk #83: Waves Makefile error

Chospanda, 2023-04-20:
I'm trying to build "waves" project on msys using "make" command and getting this error

Compiling _unit.c
Compiling waves.cpp
Linking /home/logue-sdk/platform/nutekt-digital/waves/build/waves.elf
c:/msys64/home/logue-sdk/tools/gcc/gcc-arm-none-eabi-5_4-2016q3/bin/../lib/gcc/arm-none-eabi/5.4.1/.
./../../../arm-none-eabi/bin/ld.exe: cannot open linker script file /home/logue-sdk/platform/nutekt-
digital/waves/ld/userosc.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
make: *** [Makefile:187: /home/logue-sdk/platform/nutekt-digital/waves/build/waves.elf] Error 1

Any suggestions?

sophialancini, 2023-04-24:
I ran across the same issue and it was solved already in [issue 7](https://github.com/korginc/logue-sdk/issues/7), you just have to run **git submodule update --init** before the compilation.

Jerem-yJohnson, 2023-05-09:
Hi there, 

I am getting the same error message as Chospanda. 

Did running "git submodule update --init" fix this problem for you @Chospanda ? It does not seem to be working for me. I am wondering if this is a separate issue. 

I will try to report back with any troubleshooting info I have uncovered.

Thank you @sophialancini for the guidance thus far.

Jeremy

Chospanda, 2023-05-09:
Thanks for the reply,

Running "git submodule update --init" does not fix this problem. Same error occurs.