korginc/logue-sdk #86: Cycling74 ~gen / RNBO example

suaw3k, 2023-05-13:
**Is your feature request related to a problem? Please describe.**
No available example for building Louge engines using Cycling74 ~gen or RNBO. 

**Describe the solution you'd like**
A simple example along with the documentation of building engines using ~gen and RNBO.

I'm not sure if there is a possibility to export a ~gen or RNBO project and use it as a sound engine for Korg. If so, is there any available documentation with a simple example?

Thanks

dromer, 2023-07-31:
@suaw3k RNBO would be unlikely due to https://forum.electro-smith.com/t/plans-to-support-rnbo/3131/6

gen~ and hvcc (the pd equivalent of rnbo) would be interesting to have. I've been looking at how the SDK works, but haven't really figured out how the process loop actually works.

suaw3k, 2023-10-29:
Thanks for explanation

closed 2023-10-29 by suaw3k