korginc/logue-sdk #89: Drumlogue: The volume of the signal routed to pre/post master

boochow, 2023-10-29:
The current architecture of the Drumlogue copies the audio signal to both the delay and reverb inputs, in addition to routing towards the pre/post master FX. Also, there is no dedicated parameter to set the signal level to the pre/post master FX. The global volume control, while present, impacts all signals universally, and thus, it cannot be utilized for this purpose. Consequently, this results in an inability to adjust the signal level to the pre/post master FX, thereby restricting the wet signal ratio to a maximum of 50%. 

So, I think an additional parameter should be in Drumlogue to adjust the signal level directed towards the pre/post master FX. 

The logue SDK 1.x, in contrast, doesn't have any route to bypass the delay/reverb section. The delay/reverb units must mix the dry and wet signals. It is an important difference between SDK1 and SDK2 and should be clearly stated in the documentation.

boochow, 2023-12-03:
@GeertKoekenbier I tested connecting NTS-1 to Drumlogue's audio input. I felt no distortion when "AUDIO IN" is directory routed to "OUTPUT." Other routings cause modification of the sound, but it is normal.

GeertKoekenbier, 2023-12-03:
Fyi .mov audio in DL crackling
[DL crackling 003.mov.zip](https://github.com/korginc/logue-sdk/files/13538364/DL.crackling.003.mov.zip)

GeertKoekenbier, 2023-12-04:
@boochow Yes that was an extreme example but no master FX were added. But i have it with all the output settings and Db variation. Today i used the seem used testing gear's output to a UNO synth PRO X audio input and there was no problem.

GeertKoekenbier, 2023-12-10:
UPDATE!!!! FIX!!!
I used a simple mono mini jack to mini jack from an anolog synth full out into the durmlogue and sounded GREAT!