korginc/logue-sdk #102: NTS-1 mkII: Send NoteOff event to OSC units whenever a key is off

boochow, 2024-04-20:
When the global parameter "Legato" is zero, NoteOn() is called for each key down. On the other hand, NoteOff() is called only when the last key is up. This causes an unmatch between the number of NoteOn and NoteOffs.

This is appropriate behavior for EGs but inappropriate for OSCs. I hope to send a NoteOff event for each key-up event so that we can implement a paraphonic oscillator.

AndrewCapon, 2024-05-28:
What I see is this:

Legato mode 0: Multiple Note Ons, Single Note Off.

Legato mode 1: Single Note On, Single Note Off.

What I don't understand is how the factory Oscs are working in Mode 1?

AndrewCapon, 2024-05-28:
So basically they work by not using NoteOn values at all:

```
inline void NoteOn(uint8_t note, uint8_t velo) {
    (void)velo;
    
    // Schedule phase reset
    state_.flags.fetch_or(State::k_flag_reset);
    
    // TODO: should we still fully rely on osc context pitch?
  }
```

```
/** Oscillator specific unit runtime context. */
  typedef struct unit_runtime_osc_context {
    int32_t  shape_lfo;      // Shape LFO signal encoded in Q31 fixed point format
    uint16_t pitch;          // Upper 8 bits: note, Lower 8 bits: inter-note fraction
    uint16_t cutoff;         // Unused. Future.
    uint16_t resonance;      // Unused. Future.
    uint8_t  amp_eg_phase;   // Unused. Future.
    uint8_t  amp_eg_state:3; // Unused. Future.
    uint8_t  padding0:5;
    unit_runtime_osc_notify_input_usage_ptr notify_input_usage;
  } unit_runtime_osc_context_t;
```

boochow, 2024-06-14:
I tested this para_saw oscillator and can confirm that firmware v1.2 has fixed this issue.
https://korginc.github.io/logue-sdk/ja/unit-index/korg_examples/

closed 2024-06-14 by boochow