korginc/logue-sdk #104: Unable to connect `logue-cli` with NTS-1 digital kit `mkII`

h1romas4, 2024-04-27:
Hello.

**Describe the bug**

Unable to connect `logue-cli` with NTS-1 digital kit `mkII`

**To Reproduce**

Windows: Error: `Logue handshake failed`

```pwsh
> .\logue-cli.exe probe -V
Version: 0.07-2b API: 1.01-0
> .\logue-cli.exe probe -l -d
  Available MIDI inputs:
    in  0: NTS-1 digital kit 1 MIDI IN 0
    in  1: NTS-1 digital kit 1 KBD/KNOB 1
    in  2: AG06/AG03-1 2

  Available MIDI ouputs:
    out 0: Microsoft GS Wavetable Synth 0
    out 1: NTS-1 digital kit 1 MIDI OUT 1
    out 2: NTS-1 digital kit 1 SOUND 2
    out 3: AG06/AG03-1 3
> .\logue-cli.exe probe -i 1 -o 2 -d
>>> { f0, 42, 50, 0, 55, f7 }
Error: Search device request timed out.
Logue handshake failed.
```

Linux (Ubuntu 22.04 LTS): Error: `Logue handshake failed`

```bash
$ logue-cli probe -V
Version: 0.07-2b API: 1.01-0
$ logue-cli probe -l
  Available MIDI inputs:
    in  0: Midi Through:Midi Through Port-0 14:0
    in  1: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 28:0
    in  2: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 28:1

  Available MIDI ouputs:
    out 0: Midi Through:Midi Through Port-0 14:0
    out 1: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 28:0
    out 2: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 28:1
$ logue-cli probe -i 2 -o 2 -d # probably -i 2 -o 2
>>> { f0, 42, 50, 0, 55, f7 }
Error: Search device request timed out.
Logue handshake failed.
```

**Additional context**

It works fine when using the KORG Kontrol Editor (Windows).

[image]

Is anyone else experiencing the same event?

john-k-walton, 2024-04-27:
haven't been many units avilable yet. still waiting on mine. I use OSX for development.

TarkanAl-Kazily, 2024-04-28:
I can reproduce the issue with the current version of `logue-cli` as well on Arch Linux

```
$ logue-cli probe -V
Version: 0.07-2b API: 1.01-0

$ logue-cli probe -d
Error: Could not find matching logue MIDI ports.

Logue interface connection failed.
  Available MIDI inputs:
    in  0: Midi Through:Midi Through Port-0 14:0
    in  1: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 20:0
    in  2: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 20:1

  Available MIDI ouputs:
    out 0: Midi Through:Midi Through Port-0 14:0
    out 1: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 20:0
    out 2: NTS-1 digital kit mkII:NTS-1 digital kit mkII NTS-1 di 20:1

(master)$ logue-cli probe -i 2 -o 2 -d
>>> { f0, 42, 50, 0, 55, f7 }
Error: Search device request timed out.
Logue handshake failed.
```

TarkanAl-Kazily, 2024-06-02:
I've created a Python based MIDI CLI librarian that supports the NTS-1 MK2 for the features in `logue-cli` for Linux users / command line diehards: https://github.com/TarkanAl-Kazily/py-logue-tools

john-k-walton, 2024-06-03:
thanks Tarkan! hope they get logue-cli for mkii up. i can't stand GUI's or the Librarian.
cli is the best interface for dev's too, we can just tack the unit load on the end of the compile script; no extra steps slowing it all down with a GUI.

h1romas4, 2024-06-09:
Thank you for your excellent work! It worked perfectly on my Ubuntu 22.04.
Now I can incorporate automatic transfer of sound units into my build. Awesome.

```bash
$ uname -a
Linux thinkpad-p14s 6.5.0-35-generic #35~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Tue May  7 09:00:52 UTC 2 x86_64 x86_64 x86_64 GNU/Linux
$ python -V
Python 3.10.12
$ ~/devel/korg/py-logue-tools/main.py -p2 --type NTS1Mk2 install -f dist/nts1mkii-rust-wave.nts1mkiiunit -m osc -s 2
Device SDK version 1.1
osc slot 2 - EMPTY
Sending program in 6 messages
Sending msg osc slot 2 - packet 0 (out of 5) contains 3573 bytes
Sending msg osc slot 2 - packet 1 (out of 5) contains 3573 bytes
Sending msg osc slot 2 - packet 2 (out of 5) contains 3573 bytes
Sending msg osc slot 2 - packet 3 (out of 5) contains 3573 bytes
Sending msg osc slot 2 - packet 4 (out of 5) contains 3573 bytes
Sending msg osc slot 2 - packet 5 (out of 5) contains 179 bytes
Success
```