korginc/logue-sdk #105: NTS-1 mkII: The velocity value is always zero

boochow, 2024-04-27:
When `__unit_callback void unit_note_on(uint8_t note, uint8_t velocity)` is called, `velocity` is always zero. 
I tested with a MIDI keyboard, and the EG responded to velocity inputs, but the OSC didn't.

While this is not a serious problem, I think it would be useful to use the velocity values as an OSC's modulation source.

john-k-walton, 2024-04-27:
good catch Boochow.

personally, velocity is important for user oscillators. I have used a Yorick LFE to remap velocity to Shape/Shift-Shape CC's. it is very important for to me articulation of user oscillators, after touch would be helpful at this point as well.

boochow, 2024-06-14:
I tested this para_saw oscillator and can confirm that firmware v1.2 has fixed this issue.
https://korginc.github.io/logue-sdk/ja/unit-index/korg_examples/

closed 2024-06-14 by boochow

boochow, 2024-06-14:
Note: I slightly modified the code to see if the velocity values are actually available.
```sig += voice[i].velo > 96 ? osc_sqrf(phase) : osc_sawf(phase);```