korginc/logue-sdk #107: NTS-1 mkII: The LED shows a wrong string value after scrolling starts

boochow, 2024-05-04:
There is a problem in displaying strings of type `k_unit_param_type_strings`.

(1)Values of 4 or less characters and values of 5 or more characters are mixed in the string returned by `getParameterStrValue()`.
(2)A value of 5 or more characters had been selected even once.
(3)A value of 4 or less characters is currently selected.

When the above conditions are met, the last selected value of 5 or more characters is displayed after scrolling starts.

```
// example: "SQUARE" is always displayed as a scrolling string.
    inline const char * getParameterStrValue(uint8_t index, int32_t value) cons\
t {
   	if (index == 2) {
	    if (value == 0) {
	        return "SINE";   // = 4 chars                                   
            } else {
	        return "SQUARE"; // > 4 chars                                   
	    }
	} else {
	    return nullptr;
	}
    }
```

AndrewCapon, 2024-06-02:
Hi @boochow 

You didn't by any chance find a clever way around this did you?

Thanks

Andy

boochow, 2024-06-02:
@AndrewCapon I made all strings to be longer than four characters :-)

AndrewCapon, 2024-06-02:
:)

Nice one, I will do the same.

boochow, 2024-06-14:
I tested firmware v1.2 and can confirm that this issue has been fixed.

closed 2024-06-14 by boochow