korginc/logue-sdk #116: NTS-1 mkII: MIDI sen/receive parameters value inconsistency

dukesrg, 2024-07-10:
The current MIDI CC/NRPN nadling implementation have inconsistencies.
Just an example for OSC A MSB/LSB handling:
1. OSC A LSB (CC102) sent before MSB (CC54) might be ignored, i.e. only the CC54 might be reflected in program dump afetr CC102/CC54 sequence. The CC54/CC102 sequence seems to work consisnent in that term. It should be fixed or reflected in MIDI implementation document similar to  *2-1
2. The valies itself are inconsistent: for the OSC A with 0....16363 range (changeing the range to 0...1023 change nothing) in the header parameter definition, sending 0...1023 in higher 10 bits of 14-bit CC54/CC102 qill result in the following OSC A valie in program dump:
- 0...512 - values are in par
- 513-1023 - values are 1 less, except of the next:
- 522 and every 8th up to 1018 are 9 less
- setting lower 4 bits of LSB (CC102) just changes the reusult just only for 1023

The result is reproducible, test result sheet available here:
https://docs.google.com/spreadsheets/d/1xZaxXdUIRua6vdT9HF-SRPVPFKR3oh125u0vGNMLQ0g/edit?usp=sharing

dukesrg, 2025-09-29:
@xiashj24 any chance these issues will be addressed in mkII & NTS-3 formware updates?

xiashj24, 2025-09-30:
@dukesrg Thank you a lot for investigating this. It sounds like a vicious bug, and totally possible because user tests are generally not able to detect such precision issues.
I was not involved in the logue-sdk runtime implementation of NTS-1 mk2 and NTS-3, but this sounds like we are doing something wrong which might even affect future products. I will try to reproduce it and find out what caused the bug.